jzolee/HAL2UDP #3: tutorial installation

sofwarestore, 2023-07-18:
i have try install but i get error this
https://drive.google.com/file/d/1lIcT0Yi34sxzRCUfmFuPRTwbWqgHbC6g/view?usp=drive_link

before i already have linuxcnc but with lpt port..so i want to add esp32 but i got error like picture..is there any suggestion

jzolee, 2023-07-21:
> i have try install but i get error this https://drive.google.com/file/d/1lIcT0Yi34sxzRCUfmFuPRTwbWqgHbC6g/view?usp=drive_link
> 
> before i already have linuxcnc but with lpt port..so i want to add esp32 but i got error like picture..is there any suggestion

I removed the confusing lines in the udp2.ini file - I hope I found them all. I'm sorry, but I uploaded my own config file, which contains useless stuff for others.

sofwarestore, 2023-07-21:
i still get error this

https://drive.google.com/file/d/1OFOUGBFV6qgJI5Y66VLqsO5w3HodtxRl/view?usp=sharing

but i already can run the program..is it all right to ignore the error

closed 2023-07-21 by sofwarestore