jzolee/HAL2UDP #4: Problem when trying to run AXIS and also to compile and write the code on the ESP32.

cidao1984, 2024-04-16:
Hello everything is fine? I followed the description step by step, compiled the files and replaced them in the folder where stepconf created the original files.
When I tried to run axis, it returned the error in this image, I didn't quite understand why it presented this error.
Did you get this error message? How do I resolve it?
[image]

To compile the code for the ESP32 on platformio, it indicated that the arduino.h file was missing, I placed this file in the same folder as the source code and it indicated that another different .h file was missing, I need to add these files to the folder of the source code to be able to compile and send the executable to the ESP32 board?

Thanks.

cidao1984, 2024-04-16:
I looked at the 'tutorial installation' topic, where you said that 'I'm sorry, but I uploaded my own config file...' and that you had removed the unnecessary lines. Even so, the above error appeared, which had also appeared for softwarestore when he created the topic.
  Could you indicate what I should remove in the udp2.ini file? Thanks.

jzolee, 2024-04-26:
Hello!

it probably helps if you comment out the following lines with #:
```
# OPEN_FILE = ../../nc_files/test.ngc
# PROGRAM_PREFIX = ../../nc_files
``` 

configuration help:
https://linuxcnc.org/docs/html/config/ini-config.html

cidao1984, 2024-04-26:
Ok I'll test it, thanks for the answer.
Did you follow any tutorial to develop how Hal sends data/signals to the ethernet port with udp protocol instead of the parallel port? I'm searching the Linuxcnc forum and wikis to try to understand, but I'm still confused about it....

cidao1984, 2024-04-26:
Was it necessary to create a driver for linuxcnc to communicate over ethernet? Similar to in the past when we bought a card for the computer and a CD or floppy disk came with it with the driver file for installation/operation...

jzolee, 2024-04-26:
the driver you mean: udp.comp (in this project)

jzolee, 2024-04-26:
https://linuxcnc.org/docs/2.4/html/hal_comp.html

cidao1984, 2024-04-27:
Ok, thanks for the information, hugs.

cidao1984, 2024-04-27:
Friend, I forgot one more question about something that happened when I tried to compile, when I ran the platform and compiled and recorded the code on the ESP32, I received some warnings that files with the .H extension were missing, I placed the indicated file in the source code folder and platformio indicated another .H file that was also missing, I did this with 4 files and it continued to indicate a missing file.
  How did you compile and record ESP32 on platformio without having this problem? Did you add any specific library that already contains all the necessary files?

jzolee, 2024-04-28:
I hope it will work like this:

1. Download ZIP
[image]
2. extract to any empty folder:
[image]
3. start VSCode -> File -> Open Folder...
[image]
4. wait patiently while platformio downloads the required components.

jzolee, 2024-04-28:
ps: the folder containing platformio.ini should be opened, not the src folder.

cidao1984, 2024-04-29:
Thank you very much, I'm studying here.

cidao1984, 2024-04-29:
I managed to record the ESP32, the Axis executed and communicated, but this latency problem message is appearing.
  Did this message appear to you too? Is it just a matter of ignoring it or do I need to do something to fix it?
[image]

jzolee, 2024-05-01:
If I start axis immediately after turning on the computer, I also get this error message, but it doesn't cause any problems.
I don't know why, but I didn't try to find out because it doesn't cause any problems.