jzolee/HAL2UDP #5: No communication

Fabriciokaka, 2024-06-12:
hello everything is fine?  I did the entire installation step by step, I can open Linux, but I cannot communicate between the computer and the card.  Could this have something to do with the UDP port?
[image]

jzolee, 2024-06-17:
hello,
the ip address of the module is 192.168.96.54
You must set the IP address on your computer to 192.168.96.xx (direct connection is required - without a router or switch)
It uses udp port 58427, however it only responds to valid messages.

jzolee, 2024-06-17:
one more thing:
if you use the "dev" branch of the source code, there is a change in the hardware:

ESP32 GPIO 39 <- W5500 INT