jzolee/HAL2UDP #8: ESPduino + CNCshield + W5500

tuxcnc, 2024-10-24:
https://www.cnc.info.pl/espduino-cncshield-w5500-czyli-najtanszy-sterownik-ethernetowy-t123714.html
Polish,  you could use Google Translator, but most important is attached archive.
Tested, seems work properly.

tuxcnc, 2024-10-24:
[HAL2UDP-CNCshield.zip](https://github.com/user-attachments/files/17511707/HAL2UDP-CNCshield.zip)

jzolee, 2024-11-07:
Dear tuxcnc! I'm glad you tried it and developed it according to your needs. I think you should try the following version:

https://github.com/jzolee/HAL2UDP/tree/dev

I haven't had time to test all parts of the code yet, but I hope it's not buggy. Movement and simple I/Os work, but I haven't tested pwm. An important change is that the INT pin of the W5500 module is also connected.