jzolee/HAL2UDP #11: PWM in 2.0 not works

tuxcnc, 2025-05-13:
Maybe I'm doing something wrong, but PWM doesn't work for me.

jzolee, 2025-05-18:
Hello, unfortunately there are limitations to using pwm:
- the frequency can only be set once, only the duty cycle can be changed during operation.
- all pwm outputs can only be operated with the same frequency. (this can be expanded to a maximum of 4 different frequencies with some software modifications)
If the error is not related to these, please provide more information on how you tried to use it.

tuxcnc, 2025-05-19:
I found following :
1. udp.ready is always true
2. outputs 0,1,2 works as outputs, not works as pwm
3. outputs 3,4,5 are always high and not works at all

jzolee, 2025-05-19:
Hello,
I modified the pwm handling of the code. Now 4 different frequencies can be used for the 6 output pins.
For example (in the xxxx.hal file):
setp udp.pwm.0.freq 1000
setp udp.pwm.0 0.5
setp udp.pwm.1.freq 2000
setp udp.pwm.1 0.5
setp udp.pwm.2.freq 3000
setp udp.pwm.2 0.5
setp udp.pwm.3.freq 4000
setp udp.pwm.3 0.5
setp udp.pwm.4.freq 1000
setp udp.pwm.4 0.5
setp udp.pwm.5.freq 2000
setp udp.pwm.5 0.5

I can't verify its operation right now, but if you want, you can try it: https://github.com/jzolee/HAL2UDP/tree/2.0.pwm.imp